smunaut/ice40-playground #11: Regarding riscv_usb simulation issue.

ghanesimit, 2020-01-09:
I have tried to run simulation for riscv_usb using $make sim
But will get following error.
iverilog -Wall -DSIM=1 -DBOARD_ICEBREAKER=1 -o /data/local/Desktop/USB_2019/RISC_V/ice40-playground/projects/riscv_usb/build-tmp/top_tb \
	-I/data/local/Desktop/USB_2019/RISC_V/ice40-playground/projects/riscv_usb/rtl -I/data/local/Desktop/USB_2019/RISC_V/ice40-playground/cores/usb/rtl/ -I/data/local/Desktop/USB_2019/RISC_V/ice40-playground/cores/misc/rtl/ -I/data/local/Desktop/USB_2019/RISC_V/ice40-playground/projects/riscv_usb/sim -I/data/local/Desktop/USB_2019/RISC_V/ice40-playground/cores/usb/sim/ -I/data/local/Desktop/USB_2019/RISC_V/ice40-playground/cores/misc/sim/ \
	-l/usr/local/share/yosys/ice40/cells_sim.v -l/data/local/Desktop/USB_2019/RISC_V/ice40-playground/cores/usb/rtl/usb.v -l/data/local/Desktop/USB_2019/RISC_V/ice40-playground/cores/usb/rtl/usb_crc.v -l/data/local/Desktop/USB_2019/RISC_V/ice40-playground/cores/usb/rtl/usb_ep_buf.v -l/data/local/Desktop/USB_2019/RISC_V/ice40-playground/cores/usb/rtl/usb_ep_status.v -l/data/local/Desktop/USB_2019/RISC_V/ice40-playground/cores/usb/rtl/usb_phy.v -l/data/local/Desktop/USB_2019/RISC_V/ice40-playground/cores/usb/rtl/usb_rx_ll.v -l/data/local/Desktop/USB_2019/RISC_V/ice40-playground/cores/usb/rtl/usb_rx_pkt.v -l/data/local/Desktop/USB_2019/RISC_V/ice40-playground/cores/usb/rtl/usb_trans.v -l/data/local/Desktop/USB_2019/RISC_V/ice40-playground/cores/usb/rtl/usb_tx_ll.v -l/data/local/Desktop/USB_2019/RISC_V/ice40-playground/cores/usb/rtl/usb_tx_pkt.v -l/data/local/Desktop/USB_2019/RISC_V/ice40-playground/cores/misc/rtl/delay.v -l/data/local/Desktop/USB_2019/RISC_V/ice40-playground/cores/misc/rtl/fifo_sync_ram.v -l/data/local/Desktop/USB_2019/RISC_V/ice40-playground/cores/misc/rtl/fifo_sync_shift.v -l/data/local/Desktop/USB_2019/RISC_V/ice40-playground/cores/misc/rtl/glitch_filter.v -l/data/local/Desktop/USB_2019/RISC_V/ice40-playground/cores/misc/rtl/ram_sdp.v -l/data/local/Desktop/USB_2019/RISC_V/ice40-playground/cores/misc/rtl/prims.v -l/data/local/Desktop/USB_2019/RISC_V/ice40-playground/cores/misc/rtl/pdm.v -l/data/local/Desktop/USB_2019/RISC_V/ice40-playground/cores/misc/rtl/pwm.v -l/data/local/Desktop/USB_2019/RISC_V/ice40-playground/cores/misc/rtl/uart_rx.v -l/data/local/Desktop/USB_2019/RISC_V/ice40-playground/cores/misc/rtl/uart_tx.v -l/data/local/Desktop/USB_2019/RISC_V/ice40-playground/cores/misc/rtl/uart_wb.v -l/data/local/Desktop/USB_2019/RISC_V/ice40-playground/cores/misc/rtl/xclk_strobe.v -l/data/local/Desktop/USB_2019/RISC_V/ice40-playground/cores/misc/rtl/xclk_wb.v -l/data/local/Desktop/USB_2019/RISC_V/ice40-playground/projects/riscv_usb/rtl/bridge.v -l/data/local/Desktop/USB_2019/RISC_V/ice40-playground/projects/riscv_usb/rtl/picorv32.v -l/data/local/Desktop/USB_2019/RISC_V/ice40-playground/projects/riscv_usb/rtl/soc_bram.v -l/data/local/Desktop/USB_2019/RISC_V/ice40-playground/projects/riscv_usb/rtl/soc_spram.v -l/data/local/Desktop/USB_2019/RISC_V/ice40-playground/projects/riscv_usb/rtl/sysmgr.v -lsim/spiflash.v -lrtl/top.v \
	sim/top_tb.v
iverilog: invalid option -- 'l'
../../build/project-rules.mk:92: recipe for target '/data/local/Desktop/USB_2019/RISC_V/ice40-playground/projects/riscv_usb/build-tmp/top_tb' failed
make: *** [/data/local/Desktop/USB_2019/RISC_V/ice40-playground/projects/riscv_usb/build-tmp/top_tb] Error 1

Will you please help.

smunaut, 2020-01-09:
What's your iverilog version ?  You need at least version 10.2

ghanesimit, 2020-01-09:
My version is (v0_9_7)
Icarus Verilog version 0.9.7  (v0_9_7)

Copyright 1998-2010 Stephen Williams

  This program is free software; you can redistribute it and/or modify
  it under the terms of the GNU General Public License as published by
  the Free Software Foundation; either version 2 of the License, or
  (at your option) any later version.

  This program is distributed in the hope that it will be useful,
  but WITHOUT ANY WARRANTY; without even the implied warranty of
  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
  GNU General Public License for more details.

  You should have received a copy of the GNU General Public License along
  with this program; if not, write to the Free Software Foundation, Inc.,
  51 Franklin Street, Fifth Floor, Boston, MA 02110-1301, USA.

iverilog: no source files.

Usage: iverilog [-ESvV] [-B base] [-c cmdfile|-f cmdfile]
                [-g1995|-g2001|-g2005] [-g<feature>]
                [-D macro[=defn]] [-I includedir] [-M depfile] [-m module]
                [-N file] [-o filename] [-p flag=value]
                [-s topmodule] [-t target] [-T min|typ|max]
                [-W class] [-y dir] [-Y suf] source_file(s)

See the man page for details.

I am using ubuntu 16.04 LTS. And using apt-get install iverilog. I am getting this version.
Will you please suggest me the download link for same.

smunaut, 2020-01-09:
That is way too old   0.9.7 was released in 2013 ...

You need to uninstall it with apt-get purge iverilog, then build one for yourself from source from :
ftp://icarus.com/pub/eda/verilog/v10/verilog-10.3.tar.gz

ghanesimit, 2020-01-09:
After installing iverilog v10.3
I am getting below error.

simit.ghane.457@CI5LUB051714:/data/local/Desktop/USB_2019/RISC_V/ice40-playground/projects/riscv_usb$ make sim 
/data/local/Desktop/USB_2019/RISC_V/ice40-playground/cores/usb/utils/microcode.py > /data/local/Desktop/USB_2019/RISC_V/ice40-playground/projects/riscv_usb/build-tmp/usb_trans_mc.hex
cp -a /data/local/Desktop/USB_2019/RISC_V/ice40-playground/cores/usb/data/usb_ep_status.hex /data/local/Desktop/USB_2019/RISC_V/ice40-playground/projects/riscv_usb/build-tmp/usb_ep_status.hex
cp fw/boot.hex /data/local/Desktop/USB_2019/RISC_V/ice40-playground/projects/riscv_usb/build-tmp/boot.hex
iverilog -Wall -DSIM=1 -DBOARD_ICEBREAKER=1 -o /data/local/Desktop/USB_2019/RISC_V/ice40-playground/projects/riscv_usb/build-tmp/top_tb \
	-I/data/local/Desktop/USB_2019/RISC_V/ice40-playground/projects/riscv_usb/rtl -I/data/local/Desktop/USB_2019/RISC_V/ice40-playground/cores/usb/rtl/ -I/data/local/Desktop/USB_2019/RISC_V/ice40-playground/cores/misc/rtl/ -I/data/local/Desktop/USB_2019/RISC_V/ice40-playground/projects/riscv_usb/sim -I/data/local/Desktop/USB_2019/RISC_V/ice40-playground/cores/usb/sim/ -I/data/local/Desktop/USB_2019/RISC_V/ice40-playground/cores/misc/sim/ \
	-l/usr/local/share/yosys/ice40/cells_sim.v -l/data/local/Desktop/USB_2019/RISC_V/ice40-playground/cores/usb/rtl/usb.v -l/data/local/Desktop/USB_2019/RISC_V/ice40-playground/cores/usb/rtl/usb_crc.v -l/data/local/Desktop/USB_2019/RISC_V/ice40-playground/cores/usb/rtl/usb_ep_buf.v -l/data/local/Desktop/USB_2019/RISC_V/ice40-playground/cores/usb/rtl/usb_ep_status.v -l/data/local/Desktop/USB_2019/RISC_V/ice40-playground/cores/usb/rtl/usb_phy.v -l/data/local/Desktop/USB_2019/RISC_V/ice40-playground/cores/usb/rtl/usb_rx_ll.v -l/data/local/Desktop/USB_2019/RISC_V/ice40-playground/cores/usb/rtl/usb_rx_pkt.v -l/data/local/Desktop/USB_2019/RISC_V/ice40-playground/cores/usb/rtl/usb_trans.v -l/data/local/Desktop/USB_2019/RISC_V/ice40-playground/cores/usb/rtl/usb_tx_ll.v -l/data/local/Desktop/USB_2019/RISC_V/ice40-playground/cores/usb/rtl/usb_tx_pkt.v -l/data/local/Desktop/USB_2019/RISC_V/ice40-playground/cores/misc/rtl/delay.v -l/data/local/Desktop/USB_2019/RISC_V/ice40-playground/cores/misc/rtl/fifo_sync_ram.v -l/data/local/Desktop/USB_2019/RISC_V/ice40-playground/cores/misc/rtl/fifo_sync_shift.v -l/data/local/Desktop/USB_2019/RISC_V/ice40-playground/cores/misc/rtl/glitch_filter.v -l/data/local/Desktop/USB_2019/RISC_V/ice40-playground/cores/misc/rtl/ram_sdp.v -l/data/local/Desktop/USB_2019/RISC_V/ice40-playground/cores/misc/rtl/prims.v -l/data/local/Desktop/USB_2019/RISC_V/ice40-playground/cores/misc/rtl/pdm.v -l/data/local/Desktop/USB_2019/RISC_V/ice40-playground/cores/misc/rtl/pwm.v -l/data/local/Desktop/USB_2019/RISC_V/ice40-playground/cores/misc/rtl/uart_rx.v -l/data/local/Desktop/USB_2019/RISC_V/ice40-playground/cores/misc/rtl/uart_tx.v -l/data/local/Desktop/USB_2019/RISC_V/ice40-playground/cores/misc/rtl/uart_wb.v -l/data/local/Desktop/USB_2019/RISC_V/ice40-playground/cores/misc/rtl/xclk_strobe.v -l/data/local/Desktop/USB_2019/RISC_V/ice40-playground/cores/misc/rtl/xclk_wb.v -l/data/local/Desktop/USB_2019/RISC_V/ice40-playground/projects/riscv_usb/rtl/bridge.v -l/data/local/Desktop/USB_2019/RISC_V/ice40-playground/projects/riscv_usb/rtl/picorv32.v -l/data/local/Desktop/USB_2019/RISC_V/ice40-playground/projects/riscv_usb/rtl/soc_bram.v -l/data/local/Desktop/USB_2019/RISC_V/ice40-playground/projects/riscv_usb/rtl/soc_spram.v -l/data/local/Desktop/USB_2019/RISC_V/ice40-playground/projects/riscv_usb/rtl/sysmgr.v -lsim/spiflash.v -lrtl/top.v \
	sim/top_tb.v
/data/local/Desktop/USB_2019/RISC_V/ice40-playground/cores/usb/rtl/usb.v:28: warning: timescale for usb inherited from another file.
/usr/local/share/yosys/ice40/cells_sim.v:1: ...: The inherited timescale is here.
/data/local/Desktop/USB_2019/RISC_V/ice40-playground/cores/usb/rtl/usb_crc.v:28: warning: timescale for usb_crc inherited from another file.
/usr/local/share/yosys/ice40/cells_sim.v:1: ...: The inherited timescale is here.
/data/local/Desktop/USB_2019/RISC_V/ice40-playground/cores/usb/rtl/usb_ep_buf.v:28: warning: timescale for usb_ep_buf inherited from another file.
/usr/local/share/yosys/ice40/cells_sim.v:1: ...: The inherited timescale is here.
/data/local/Desktop/USB_2019/RISC_V/ice40-playground/cores/usb/rtl/usb_ep_status.v:28: warning: timescale for usb_ep_status inherited from another file.
/usr/local/share/yosys/ice40/cells_sim.v:1: ...: The inherited timescale is here.
/data/local/Desktop/USB_2019/RISC_V/ice40-playground/cores/usb/rtl/usb_phy.v:28: warning: timescale for usb_phy inherited from another file.
/usr/local/share/yosys/ice40/cells_sim.v:1: ...: The inherited timescale is here.
/data/local/Desktop/USB_2019/RISC_V/ice40-playground/cores/usb/rtl/usb_rx_ll.v:28: warning: timescale for usb_rx_ll inherited from another file.
/usr/local/share/yosys/ice40/cells_sim.v:1: ...: The inherited timescale is here.
/data/local/Desktop/USB_2019/RISC_V/ice40-playground/cores/usb/rtl/usb_rx_pkt.v:28: warning: timescale for usb_rx_pkt inherited from another file.
/usr/local/share/yosys/ice40/cells_sim.v:1: ...: The inherited timescale is here.
/data/local/Desktop/USB_2019/RISC_V/ice40-playground/cores/usb/rtl/usb_trans.v:28: warning: timescale for usb_trans inherited from another file.
/usr/local/share/yosys/ice40/cells_sim.v:1: ...: The inherited timescale is here.
[... cut ...]
/data/local/Desktop/USB_2019/RISC_V/ice40-playground/projects/riscv_usb/rtl/bridge.v:36: warning: timescale for bridge inherited from another file.
/usr/local/share/yosys/ice40/cells_sim.v:1: ...: The inherited timescale is here.
/data/local/Desktop/USB_2019/RISC_V/ice40-playground/projects/riscv_usb/rtl/bridge.v:42: syntax error
I give up.
../../build/project-rules.mk:92: recipe for target '/data/local/Desktop/USB_2019/RISC_V/ice40-playground/projects/riscv_usb/build-tmp/top_tb' failed
make: *** [/data/local/Desktop/USB_2019/RISC_V/ice40-playground/projects/riscv_usb/build-tmp/top_tb] Error 2

ghanesimit, 2020-01-09:
To solve above error I've made following changes.

simit.ghane.457@CI5LUB051714:/data/local/Desktop/USB_2019/RISC_V/ice40-playground/projects/riscv_usb$ git diff rtl/bridge.v
diff --git a/projects/riscv_usb/rtl/bridge.v b/projects/riscv_usb/rtl/bridge.v
index d7fb04a..0dc48a1 100644
--- a/projects/riscv_usb/rtl/bridge.v
+++ b/projects/riscv_usb/rtl/bridge.v
@@ -38,7 +38,7 @@ module bridge #(
        parameter integer WB_DW = 32,
        parameter integer WB_AW = 16,
        parameter integer WB_AI =  2,
-       parameter integer WB_REG = 0,   // [0] = cyc / [1] = addr/wdata/wstrb / [2] = ack/rdata
+       parameter integer WB_REG = 0    // [0] = cyc / [1] = addr/wdata/wstrb / [2] = ack/rdata
 )(
        /* PicoRV32 bus */
        input  wire [31:0] pb_addr,
simit.ghane.457@CI5LUB051714:/data/local/Desktop/USB_2019/RISC_V/ice40-playground/projects/riscv_usb$ git diff rtl/top.v
diff --git a/projects/riscv_usb/rtl/top.v b/projects/riscv_usb/rtl/top.v
index a187652..1ee3a90 100644
--- a/projects/riscv_usb/rtl/top.v
+++ b/projects/riscv_usb/rtl/top.v
@@ -464,8 +464,8 @@ module top (
        assign wb_rdata[4][31:16] = 16'h0000;
 
        // EP buffer interface
-       always @(posedge clk_24m)
-               wb_ack[5] <= wb_cyc[5] & ~wb_ack[5];
+       //always @(posedge clk_24m)
+       assign wb_ack[5] = wb_cyc[5] & ~wb_ack[5];
 
        assign ep_tx_addr_0 = wb_addr[8:0];
        assign ep_tx_data_0 = wb_wdata;

Will you please confirm is it ok or not?

smunaut, 2020-01-09:
The first one is fine, the second one is not, you can't just remove the the always and make it combinatorial ....

ghanesimit, 2020-01-09:
If I am not modifying the have the following error .

rtl/top.v:468: error: wb_ack['sd5] is not a valid l-value in top_tb.dut_I.
rtl/top.v:103:      : wb_ack['sd5] is declared here as wire.
1 error(s) during elaboration.

smunaut, 2020-01-09:
You should use the 'usb' branch of this repo, it has the latest USB related code in it, including a fix for that issue.

ghanesimit, 2020-01-09:
I have done that too.

using following 

$git checkout remotes/origin/usb
$make clean
$make sim
make: *** No rule to make target '/data/local/Desktop/USB_2019/RISC_V/ice40-playground/projects/riscv_usb/build-tmp/dfu_helper_tb', needed by 'sim'.  Stop.

smunaut, 2020-01-09:
Just remove https://github.com/smunaut/ice40-playground/blob/usb/projects/riscv_usb/Makefile#L18

smunaut, 2020-01-09:
But tbh the system level simulations don't do anything anyway ... I use them during development to test early stuff but once it works I don't keep them up to date so it won't really do anything anyway.

smunaut, 2020-01-09:
Anyway, this is really not the proper channel for this, so closing the issue. I'm on the usb discord  https://discord.gg/HKAhHub  or on ##openfpga channel on discord for this kind of stuff.

closed 2020-01-09 by smunaut