smunaut/ice40-playground #13: Can't build the rgb_panel project

Taki-B, 2020-04-17:
I've set up my FPGA toolchain using Esden's smart shell script (summon-fpga-tools.sh) but can't get the project done. Yosys tries 

```
38.47.1. Executing Verilog-2005 frontend: /home/example/sft/bin/../share/yosys/ice40/cells_map.v
Parsing Verilog input from `/home/example/sft/bin/../share/yosys/ice40/cells_map.v' to AST representation.
Generating RTLIL representation for module `\$_DFF_N_'.
Generating RTLIL representation for module `\$_DFF_P_'.
Generating RTLIL representation for module `\$_DFFE_NN_'.
Generating RTLIL representation for module `\$_DFFE_PN_'.
Generating RTLIL representation for module `\$_DFFE_NP_'.
Generating RTLIL representation for module `\$_DFFE_PP_'.
Generating RTLIL representation for module `\$_DFF_NN0_'.
Generating RTLIL representation for module `\$_DFF_NN1_'.
Generating RTLIL representation for module `\$_DFF_PN0_'.
Generating RTLIL representation for module `\$_DFF_PN1_'.
Generating RTLIL representation for module `\$_DFF_NP0_'.
Generating RTLIL representation for module `\$_DFF_NP1_'.
Generating RTLIL representation for module `\$_DFF_PP0_'.
Generating RTLIL representation for module `\$_DFF_PP1_'.
Generating RTLIL representation for module `\$__DFFE_NN0'.
Generating RTLIL representation for module `\$__DFFE_NN1'.
Generating RTLIL representation for module `\$__DFFE_PN0'.
Generating RTLIL representation for module `\$__DFFE_PN1'.
Generating RTLIL representation for module `\$__DFFE_NP0'.
Generating RTLIL representation for module `\$__DFFE_NP1'.
Generating RTLIL representation for module `\$__DFFE_PP0'.
Generating RTLIL representation for module `\$__DFFE_PP1'.
Successfully finished Verilog frontend.

38.47.2. Continuing TECHMAP pass.
No more expansions possible.
Removed 32 unused cells and 670 unused wires.

38.48. Executing OPT_LUT pass (optimize LUTs).
```

and then stops with the error 

```
terminate called after throwing an instance of 'std::out_of_range'
  what():  dict::at()
make: *** [../../build/project-rules.mk:75: /home/example/ice40-playground/projects/rgb_panel/build-tmp/rgb_panel.json]
```

I've tried to update the toolchain but the result is the same. Unfortunately, I'm a beginner and do not understand what this error means and googling around didn't help.

Could someone who could builf the project post his toolchain version? I'm using

```
Yosys 0.9+2406 (git sha1 UNKNOWN, clang 7.0.1 -fPIC -Os)
iverilog v10_3
```

and the newest nextpnr-ice40, icepack and iceprog.

The toolchain itself seems to work at least partialy, as I can build the icebraker-workshop project or some the examples provided with BlackIce II.

Thanks in advance!

smunaut, 2020-04-17:
That's yosys crashing ... it should not ever do that, so that should be reported as a yosys bug.

It builds fine for me here, so that might be due to having yosys built using clang (I use gcc). But in anycase, the bug lies with yosys.

You should run it in gdb to get a backtrace and open an issue in yosys.

closed 2020-04-17 by smunaut

Taki-B, 2020-04-17:
Thanks for the hint. I'll try to build a gcc based yosys and then see if I can catch the bug and report it. Thanks for your time.

smunaut, 2020-04-17:
I just tried on another machine with a fresh yosys built with the summon tools and it worked fine as well.

```
Yosys 0.9+2406 (git sha1 13f4015, clang 9.0.0-2 -fPIC -Os)
```

Taki-B, 2020-04-17:
Thanks for the information. It must be then an error introduced when compiling yosys on my machine because I get the same error with the gcc version.