smunaut/ice40-playground #15: Will the codes work with ICEstudio environment?

Hazeline2018, 2020-05-26:
Hi Smunaut,

I'm new to ICE FPGA environment. And I learned that there are quite a few open FPGA platform that can be working with. I'm planning on using ICEstudio with your example project. Will that work? Would you please point me some resource on how to set up icestudio toolchain? Your help is greatly appreciated!

Thanks,
Terry

smunaut, 2020-05-26:
The toolchain used in icestudio is the classic yosys/nextpnr so it should be capable of building most of the code. However :
 - At least historically they tended to be rather outdated toolchain version while OTOH, my code tends to depends on very new versions / features of the toolchain so not all code might work with old toolchains.
 - If you want the block to show up as blocks in icestudio, you'll need some sort of magic / wrapper / block description for them to show up and you will have to replicate all the dependencies between the various verilog files yourself, this repo has a whole build system and dependency system for build that wouldn't work in icestudio IDE.  And for that part you're on your own, I never used icestudio ...

Hazeline2018, 2020-05-27:
Thanks for the prompt response. To get started easily, I can follow your original toolchain setup and get it going first. Then I can start transition to icestudio. Would you please show me how to set up the whole toolchain and environment that you project is using? Thanks!

smunaut, 2020-05-27:
There are many tutorials online. Refer to the fomu toolchain setup, the 1bitsquared workshop setup and/or ask on the 1bitsquared discord ( https://1bitsquared.com/pages/chat )

closed 2020-05-27 by smunaut

Hazeline2018, 2020-05-27:
thanks a lot! I'll dig in!