smunaut/ice40-playground #20: usb_audio firmware stops running in usb_init

Barabas5532, 2020-12-29:
I'm trying to test the usb_audio project using an iCEBreaker board. I've got the riscv_usb project running fine (with a modified pcf to match the pins in usb_audio). The PC detects it when using the d and c commands through UART.

```
[Dec29 18:51] usb 3-2.4: new full-speed USB device number 68 using xhci_hcd
[  +0.116515] usb 3-2.4: New USB device found, idVendor=1d50, idProduct=6147, bcdDevice= 0.01
[  +0.000002] usb 3-2.4: New USB device strings: Mfr=2, Product=3, SerialNumber=1
[  +0.000001] usb 3-2.4: Product: iCEBreaker
[  +0.000001] usb 3-2.4: Manufacturer: osmocom
[  +0.000000] usb 3-2.4: SerialNumber: e46998d243581a2f
[  +0.043020] cdc_acm 3-2.4:1.0: ttyACM0: USB ACM device
```

However, the usb_audio project doesn't reach the main loop. I've added extra prints to the no2usb code, and it seems like the code stops executing after the call to memset. 

```
void
usb_init(const struct usb_stack_descriptors *stack_desc)
{
	/* Main state reset */
    puts("memset start\n"); // this one prints
    memset(&g_usb, 0x00, sizeof(g_usb));
    puts("memset done\n"); // this one never prints
```

I've installed the risc-v toolchain using `make -j$(nproc) build-riscv32i-tools` of the [picorv32](https://github.com/cliffordwolf/picorv32#building-a-pure-rv32i-toolchain) repo. Could this issue be caused by having different toolchains?

Barabas5532, 2020-12-30:
Never mind, it just started working randomly.

closed 2020-12-30 by Barabas5532

smunaut, 2020-12-30:
MMm, that's ... weird.

The toolchain shouldn't be an issue since the Makefile properly selects the architecture to be plain `rv32i`.