smunaut/ice40-playground #21: rgb_panel project built successfully but nothing shows on the panel

Hazeline2018, 2021-01-15:
I used the exactly same setup as the IceBreaker ICE40 dev board with 64x64 panel. I was able to build rgb_panel project with Pattern mode successfully, but nothing showed up on the panel. it's pure blank. I read through the RTL, didn't see anywhere specifies the scan mode, eg 1/32 or 1/16. The only parameters I can see as follows in "top.v":

	// Params
`ifdef BOARD_ICEBREAKER_SINGLE2X
	localparam integer N_BANKS  = 4;
`else
	localparam integer N_BANKS  = 2;
`endif
	localparam integer N_ROWS   = 32;
	localparam integer N_COLS   = 64;
	localparam integer N_CHANS  = 3;
	localparam integer N_PLANES = 10;
	localparam integer BITDEPTH = 16;

	localparam integer LOG_N_BANKS = $clog2(N_BANKS);
	localparam integer LOG_N_ROWS  = $clog2(N_ROWS);
	localparam integer LOG_N_COLS  = $clog2(N_COLS);

Since I'm using 64x64 panel, should I change N_ROWS to 64? Tried, but still nothing shows. I also noticed that in the "pgen.v" module, these paras are hard-coded:

module pgen #(
	parameter integer N_ROWS   = 64,	// # of rows (must be power of 2!!!)
	parameter integer N_COLS   = 64,	// # of columns
	parameter integer BITDEPTH = 24,

	// Auto-set
	parameter integer LOG_N_ROWS  = $clog2(N_ROWS),
	parameter integer LOG_N_COLS  = $clog2(N_COLS)
)(

I tried other simple examples from other projects, the panel light up just fine, altering color blocks to test the panel and setup fine. Please any help would be highly appreciated!

smunaut, 2021-01-15:
64x64 panels are still 1:32 and that's specified using `N_ROWS` (which is the number of multiplexed rows). But it pushes 2 rows at once (upper and lower part of the panels) which is called `N_BANKS` here.  So the default config is good for 64x64 panels.

A few questions :
 * What kind of panels do you have ? (especially which driver chip)
 * Are you sure the connections are right ? It's easy to plug something wrong ... either feeding the panel output, or having the header the wrong way around ...

Hazeline2018, 2021-01-15:
Thanks so much for your prompt response!
I bought the 64x64 panel from Crowdsupply along with the iCEBreaker FPGA board. I checked the panel driver ,it's ICN2037 shift driver.
Yes, I'm quite sure the connections are good. I verified quite a few times. I even tested with another very simple project to have the entire panel alter R/G/B blocks which works fine.

Hazeline2018, 2021-01-15:
I checked the panel driving logic, it is 1/32 scan. The row driver is GM7258 and col driver is ICN2037 which are all standard LED panel drivers.

smunaut, 2021-01-16:
Can you try the attached bitstream ? (to exclude any build or toolchain issue)

[rgb_panel.zip](https://github.com/smunaut/ice40-playground/files/5824192/rgb_panel.zip)

Hazeline2018, 2021-01-17:
An update, I have managed to get the project working by trying an older version of hub75_top.v although didn't really figured out what was going wrong with earlier attempts. I then resyn the whole sandbox and tried again and everything works fine now. So I moved on. :-)

However, when I'm trying the video mode, I was able to flash in my own video contents (over 30s long). But when playing back, it's playing backwards, like rewinding. Tried the up/down button function, that works as expected. I guess it might be due to wrong frame counter calculation? Since the iCEBreaker FPGA board I use has on-board 16MB flash, which is large enough to fit in mins long video content, so it's not the memory overflow/wrap-around issue for sure. Any suggestions? thanks!

smunaut, 2021-01-17:
So it works with the latest version ?  Or did you have to stay on the older one ?

For the play direction, yeah look like I might have swapped them ...
Look in `vgen.v` and change :

` frame_sel <= cnt_rep_last ? 2'b10 : 2'b00;`
to
` frame_sel <= cnt_rep_last ? 2'b11 : 2'b00;`

Hazeline2018, 2021-01-17:
The latest repo works. 

Okay, will try your suggestion and update soon. Thanks a lot!

Hazeline2018, 2021-01-17:
Yep, that swapping direction fixes the problem! Thanks a lot for the great help!

I have a few other questions though and like to discuss with you further. But I'll create another thread so you can close this one if you like. Thanks again!

smunaut, 2021-01-17:
You should instead join the 1bitsquared discord channel to discuss things. ( https://1bitsquared.com/pages/chat ) or alternatively the ##openfpga  IRC channel  on freenode.

Trying to keep the issues here for actual bugs.

closed 2021-01-17 by smunaut

Hazeline2018, 2021-01-17:
all right, I'll join !