smunaut/ice40-playground #22: Various LED matrix panel support?

Hazeline2018, 2021-01-17:
Hi Sylvain,

Do you have a list of the panel driver ICs supported? I know it's working fine with ICN2037/ICN2012. Is ICN2038 or ICN2038S supported? It'll be nice to have a short list of the major driver ICs supported.

I'm custom designing my own LED matrix panels with ICN2038S and RT5958 combination. And working on expand the functions as below based on your work which I think the most completed and up-to-date FPGA based solution. My ultimate goal might be porting into Xilinx (Spartan 6 9XL) or Altera FPGA for more cost-efficiency.

1. support multiple hub75e ports (up to 6~8) to drive larger panels. The one I'm designing is 384x128 pixels;
2. support arbitrary scan schemes from 1:1 to 1:32;
3. use faster clock to have higher fresh rate, static image display >1khz, video frame rate >20fps;
4. use pwm to control the brightness on the fly;

it'll be great to have your support and help! Best!

smunaut, 2021-01-17:
I don't have any list of supported drivers, too many of them and I personally only have ICN2037 and FM6126 panels. Also it's not only dependent on the drivers ... some panels with the same driver chips have different architecture and need to be driven differently ...  But most can be made to work with small tweaks as long as (1) datasheet is available and (2) they are the classic "shift and latch" type of data panels.

I'd recommend looking into the ECP5 12F rather than going Altera or Xilinx, they're cheap, have plenty of pins, definitely enough logic to drive panels and supported by the open-tools without having to mess with ISE or Quartus.

As for the other questions :

1. The HUB75 core can be configured for an arbitrary number of PHY in parallel, see the `PHY_N` parameter.
2. The `N_ROWS` parameter sets the multiplexing used by the panels already.
3. The clock here is fully dictated by the panels ... Most can't run above 30 MHz which is what's used in the demo project. How often the panels end up being fully redrawn is a trade-off of several parameters, There is a `hub75_timing.py` utility in `cores/no2hub75/sw` that can help estimate this and play with the various parameters to see the result. 
4. You can't use PWM with those panels, they are driven using BCM.

Hazeline2018, 2021-01-18:
thanks for the advise!

regarding 2, the N_ROWS parameter seems having to be power of 2. If I want to do, say 1:20 multiplexing, what should I do to make the codes work?

smunaut, 2021-01-18:
Oh yeah, it needs to be a power of 2 ... Are there panels that have non-number of 2 multiplexing ?!?!? 

I mean you could probably avoid the scan of the 12 last row in that example but the addressing (in the frame buffer) would round up to the next power of 2. (Same as with n_cols, the addressing has a fixed number of bits in the LSB).

Hazeline2018, 2021-01-18:
Yeah, I designed a few customized LED panels with all sorts of multiplexing numbers from 1 to 32, such as 11, 20, 24, etc. Even the col numbers are arbitrary. But I guess the simple solution to odd number (i.e. not power of 2) of multiplexing could be scanning through empty rows while holding the OE high. The only problem with that is wasting fresh rate, especially for the odd numbers that far from the nearest power of 2 number. So I'm trying to tweak the codes to really support any arbitrary scan mode  w/o sacrificing fresh rate.

Hazeline2018, 2021-01-19:
BCM is used to control the brightness of the display. But which parameter can I dial to dim or bright the display? I tried a couple and that made the fresh rate noticeably slow.

Hazeline2018, 2021-01-23:
Hi Sylvain,

What parameters should I work with to dim the display? thanks!

smunaut, 2021-02-01:
As I said, go to discord for that kind of question.

closed 2021-02-01 by smunaut