smunaut/ice40-playground #24: riscv_usb crash on reboot

cbaechler, 2021-08-03:
Hi Sylvain, 

I am trying to get the `riscv_usb` example to work. So far I can build it successfully, program to the bitstream and firmware to the icebreaker and also get the console output. Furthermore I have wired the USB pins according to the description, and the usb device connects after sending command `c`! (BTW: I did not yet fully understand the pull-up to DP, what is the reason behind?) 

```
abc123Booting App image..
Flash Manufacturer : ef 40 18
Flash Unique ID    : e4 60 81 11 17 34 05 39
Command> c
Command> p
Stack:
        State: 5
HW:
        SR   : 90a2
        Tick : 05d1

EP0 OUT
        S     0026
        BD0.0 0000
        BD0.1 0000
        BD1.0 4040
        BD1.1 0040

EP0 IN
        S     0006
        BD0.0 0000
        BD0.1 0000
        BD1.0 0000
        BD1.1 0000

Data:
00 00 00 00  00 00 00 00  00 00 00 00  00 00 00 00
```

It is also visible in dfu-util: 
```
$> dfu-util -l

dfu-util 0.10

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2020 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Found Runtime: [1d50:6147] ver=0001, devnum=42, cfg=1, intf=2, path="1-3", alt=0, name="DFU runtime", serial="e460811117340539"
```

Now if I execute `make dfuprog`, send the `b` command in the console or press the button, the device seems to crash. Even the `CDONE` LED turns off. What did I miss? Any hint where I could dig further?

smunaut, 2021-08-03:
The pull-up is just part of the usb specs. That's how the hosts detects a device attaching, either DP or DN gets pulled up by a 1.5k resistor (DP for full-speed, DN for low-speed devices).

The 'b' command and the dfu runtime 'detach' implementation use the `SB_WARMBOOT` block, but for that to work, the fpga needs to have been programmed for multi-boot support which won't be the case if you just do `make prog`.  That's usually only setup if you load the firmware via DFU to a device that's been pre-prepared with a bootloader image.

Note that the `riscv_usb` demo in that repo doesn't really implement DFU it's just a demo of exposing the runtime interface of DFU. If you want the actual DFU bootloader you should be looking in https://github.com/no2fpga/no2bootloader/  and this repo also contains all the required stuff to make a proper multi-boot image.

cbaechler, 2021-08-05:
Thanks a lot for your help. 

> The pull-up is just part of the usb specs. That's how the hosts detects a device attaching, either DP or DN gets pulled up by a 1.5k resistor (DP for full-speed, DN for low-speed devices).

I just learned something about usb specs! Great :smiley: 

> If you want the actual DFU bootloader you should be looking in https://github.com/no2fpga/no2bootloader/ and this repo also contains all the required stuff to make a proper multi-boot image.

Thanks for pointing me to this repo. It took me a while to make it work, but now I can dfu-load the `riscv_usb` demo with success!

closed 2021-08-05 by cbaechler